eggsampler/ESP32-S2-Breakout #1: External crystal 32kHz

fabiuz7, 2020-04-11:
Hi Isaac,
thanks for sharing your breakout board.
I was wondering why you inserted an external 32kHz crystal when you should be able to use the internal 31.25kHz derived from 8MHz clock. Is it to improve the accuracy of the timer?

eggsampler, 2020-04-11:
Hi, thanks for the question!

Essentially yes, internal RC oscillators for most MCU's are not super accurate.

My experience with the original ESP32 and using the internal oscillator for ULP/RTC deep sleep and other timing functions showed that it drifted up to a few seconds every minute. The RC oscillators seem to vary quite a bit with temperature and also between production runs. This is probably fine for most things, but when you have deployed devices operating for weeks to months at a time remotely, the drift is very noticeable.

Some of my use cases for the new ESP32-S2 involve utilising the new ULP co-processor. With an actual RISC core instead of the older 16bit state machine this should give a lot more functionality without needing to wake up the main core for power savings. So more accurate timings using an accurate crystal that's measured like 10 to 20 PPM as opposed to a few percent makes sense.

fabiuz7, 2020-04-11:
Thanks for the clear explanation! I probably will rely on the internal clock saving few components, my application will need to sleep for some minutes between a measure and the next one, drifting off a few seconds doesn't really matter.

eggsampler, 2020-08-24:
Hi,

I have not properly tested this board yet post assembly as I am waiting for the 4.2 release of the IDF with full support for the RISC & state machine ULP. See: https://www.esp32.com/viewtopic.php?f=10&t=14532

The crystal I chose was this one: https://au.mouser.com/ProductDetail/732-FC-13532.76KAA0
Had an ESR of less than 70kR as per the datasheet, with a load capacitance of 12.5pF

I didn't use any biasing resistors (populated the series resistors with 0 and left the bias parallel resistor empty).

I only had 12pF C0G capacitors on hand, realistically I believe I should have populated these with at least 15pF to 22pF
https://www.nxp.com/docs/en/supporting-information/R_10077.pdf

Hope this helps!

fabiuz7, 2020-10-04:
I can confirm that it works with 18pf.

[Here](https://easyeda.com/fabiano.riccardi/esp32-s2-breakout-board) the board I'm working on.

closed 2020-10-04 by fabiuz7

mrWheel, 2024-02-13:
Will this board “work” without the external crystal circuitry as the chip already has a 32kHZ crystal on board?

Are you still developing this board?

eggsampler, 2024-02-13:
You will only need an 32.768khz crystal if you want an accurate clock for RTC. Otherwise it will use an internal RC oscillator tuned from the 40MHz crystal if I recall.

Have used a couple of these for small projects but nothing serious. Don't plan on doing much with it.