eggsampler/ESP32-S2-Breakout #4: Is it possible to use the 32kHz crystal with the wires?

andershedberg, 2021-03-21:
Hi, juts wondering if the 32KHz crystal has been tested and works with the layout? When I worked with STM controllers they are very sensitive to the wiring of the 32 kHz x-tal requiring grounded guard rails etc. 

Any other news if this board is working well?

eggsampler, 2021-05-27:
Hi, late reply I know, sorry!

I've just had some of the v2 boards assembled and all tests well, I've included the BOM file I used for ordering from JLCPCB which includes the specific LCSC product code for components I have used. The crystal seems fine so far, along with some series 0 ohm resistors (which probably aren't necessary, I just included them for a footprint if it needed to be tuned), no parallel resistor, and the 20f caps. Have not put it on a scope to check the waveform, but I will look at doing that soon.

There are a few boards spare, so if you want to email me via esp32 at eggsampler dot com I'd be happy to mail you a board free to try yourself.

andershedberg, 2021-05-31:
Great! I actually modified your board to fit a touch-application and will assembly it now to see if it worked! If not, I might pick you up on your kind offer!

closed 2021-05-31 by andershedberg