eggsampler/ESP32-S2-Breakout #8: What is the value of R1?

crgarcia12, 2021-09-24:
What should be the value of R1? 
I see it is marked as N/C, and I am not sure if that resistor should be there at all, or if it has to be ommited

Thanks!

eggsampler, 2021-09-25:
The crystal and associated caps and resistors form part of a Pierce Oscillator https://en.wikipedia.org/wiki/Pierce_oscillator

As seen in the ESP32 S2 Hardware Design Guidelines section '2.4.2 RTC (optional)' https://www.espressif.com/sites/default/files/documentation/esp32-s2_hardware_design_guidelines_en.pdf

Generally speaking you won't need to populate it, or even add a footprint for the series or parallel resistors there at all.

crgarcia12, 2021-09-30:
thanks a lot for answering @eggsampler!!

closed 2021-10-10 by eggsampler