eggsampler/ESP32-S2-Breakout #12: Information about the card

hamzadenizyilmaz, 2023-09-23:
Hello, I produced this card and it came to my hand, since you wrote "N / C" on the card "R1" numbered resistor "N / C", I did not put that resistor on the card, will there be a problem or not? 

Thank you good work 
[image]

eggsampler, 2023-09-23:
Hi, this resistor is not necessarily required, depending on the crystal or oscillator that is being used. Chances are you won't have a problem, especially if you're not using any RTC or clock features to even use that crystal in the first place.

https://en.wikipedia.org/wiki/Pierce_oscillator#Biasing_resistor

hamzadenizyilmaz, 2023-09-23:
> Hi, this resistor is not necessarily required, depending on the crystal or oscillator that is being used. Chances are you won't have a problem, especially if you're not using any RTC or clock [...]

Ok, thanks, I won't use RTC or clock features anyway, but I still want to ask, if I want to use these two features, which resistor would be better to put in the middle?

eggsampler, 2023-09-23:
Even if you use those features, you probably won't need any resistor there at all. The footprint is only included just in case there is any problems with the crystal.

If you notice the crystal not starting up properly, or not stabilising, or you have issues and the waveform across the crystal looks weird, you might need to put like a 10 megaohm resistor there. It is extremely unlikely this will happen though.

hamzadenizyilmaz, 2023-09-23:
Hmm I understand sir, thank you, I would like to ask you a few questions in private, sir, can I ask for your e-mail address?

hamzadenizyilmaz, 2023-09-23:
> Even if you use those features, you probably won't need any resistor there at all. The footprint is only included just in case there is any problems with the crystal.
>
> If you notice the crystal not [...]

Hmm I understand sir, thank you, I would like to ask you a few questions in private, sir, can I ask for your e-mail address?